tudelft-iv/multi_sensor_calibration #1: Package Configuration "radar_msgs" missing

Hjorth2, 2019-08-22:
Hi,

I'm trying to install the package using your guide. When I get to the 'catkin_make' I get the following error:

```
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
    Could not find a package configuration file provided by "radar_msgs" with any of the following names:

        radar_msgsConfig.cmake
        radar_msgs-config.cmake

    Add the installation prefix of "radar_msgs" to CMAKE_PREFIX_PATH or set "radar_msgs_DIR" to a directory containing one of the above files. If "radar_msgs" provides a seperate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
    multi_sensor_calibration/radar_detector/CMakeLists.txt:6 (find_package)

-- Configuring incomplete, errors occurred!
Makefile:320: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```

Is this a problem you have encountered before and is there a solution to it?

Thanks.

RonaldEnsing, 2019-08-22:
Have you installed the `radar_msgs` package (http://wiki.ros.org/radar_msgs)?

Hjorth2, 2019-08-22:
Thank you for the quick answer. That was of course the problem.

closed 2019-08-22 by Hjorth2