tudelft-iv/multi_sensor_calibration #2: Trihedral Corner Reflectors alternatives

onurgurisik, 2019-09-02:
I am searching a trihedral corner reflectors in the internet. Do you have any recommendations that you can give me about size and quality.

If you sent me a  purchase link, I would be appreciated.

I find this in the amazon. What is your thoughts about it ? 

[https://www.amazon.com/Davis-Emergency-Deluxe-Radar-Reflector/dp/B002MJKNPY/ref=sr_1_1?keywords=Radar+Reflector&qid=1567410793&rnid=2941120011&s=electronics&sr=1-1]

frankeverdij, 2019-09-02:
Hello onurgurisik,

We make our own trihedral corner reflectors from single-sided copper-plated blank circuit boards. We cut them in triangles and solder them together, see picture:

[image]

The dimensions of the corner reflector is chosen so that the RCS value is 1, 10 or 100 m^2 . A formula for calculating the RCS value from a trihedral corner reflector with known dimensions is shown at the bottom of this page: http://www.radartutorial.eu/17.bauteile/bt47.en.html

ishfaq-gist, 2019-12-03:
Hi,
Can you please tell me the exact dimension of your your trihedral corner reflector and which radar sensor you are using for this project. And the thickness Styrofoam please.

frankeverdij, 2019-12-04:
The trihedral reflector ribs are 14.0 cm. The outer edges of the reflector are size sqrt(2)*14.0 = 19.7 cm.
We use styrofoam of 6 cm thickness, but that is primarily to ensure having a straight target when standing upright.
As for the radar sensor, we use radar systems from Continental, most notably the ARS 309 and newer models.

ishfaq-gist, 2019-12-04:
thank you for your quickly reply. Actually we are using different radar sensor (Delphi) ESR, so we need to change the measurement of trihedral or any change in code. Thank you once again.

closed 2019-12-04 by frankeverdij

dejongyeong, 2020-10-07:
@ishfaq-gist 
A quick question here on the change of measurement of the trihedral or change in code, what measurement of the trihedral did you changed to or which block of code did you changed?

Thanks and looking forward to hear from you.

ishfaq-gist, 2020-10-08:
@dejongyeong 
Actually, I used the same trihedral measurement as mentioned above. I did not change anything in code.

frankeverdij, 2025-12-02:
> (tudelft-iv/multi_sensor_calibration#2)
>
> Hi @frankeverdij <https://github.com/frankeverdij> , I would like to seek
> your help in my project.
>
> GHz short-range radar
>
> My name is Harshitha, and I am working on a research project that involves
> calibrating a 77 GHz automotive radar with a camera.
>
> Our setup:
>
> Radar: Continental SRR308-21 (77 GHz FMCW short-range radar)
>
> Application: Research on radar–camera calibration for an autonomous rail
> vehicle concept (MONOCAB)
>
> Use case for trihedral: Static calibration target placed in front of the
> radar
>
> Typical distances: Reflector will usually be placed at about 5–30 meters
> from the radar
>
> Environment:some indoor lab tests
>
> We plan to use a trihedral corner reflector as a reference target and want
> it to:
>
> Provide a strong, stable radar return at 77 GHz,
>
> Behave as a point-like target (not a large extended reflector),
>
> Be reasonably compact and easy to mount and transport.
>
> Since you have experience with corner reflectors, I would like to ask for
> your recommendation:
>
> For a 77 GHz short-range radar at 5–30 m, what trihedral edge length /
> physical size would you suggest so that the reflector gives a clear, strong
> response but is not unnecessarily large?How doe we know appropriate size of
> reflector how to find it.
>
> What RCS range at or near 77 GHz would you consider appropriate for this
> kind of automotive radar calibration (short-range, research use)?
>
> Our goal is to select a reflector that is “good enough but not oversized”
> for short-range automotive radar experiments.
>
> Thank you very much in advance for your advice and suggestions.

Hello Harshitha,

It has been 7 years since i last worked on radar systems and i am not an
expert in this matter. The webpage we usually consult for calculation of
radar reflection is https://www.radartutorial.eu/17.bauteile/bt47.en.html
My coworkers have published a paper in 2019 which is
https://pure.tudelft.nl/ws/portalfiles/portal/110958083/An_Extrinsic_Calibration_Tool_for_Radar_Camera_and_Lidar.pdf
I have only skimmed it, but it should have a description of the radar
reflector used in this calibration setup. I remember it to be about 5-6 cm
in corner length.
Any RCS range which the radar can detect is good enough. Just make a few
reflectors with varying size and check which reflector still gives a decent
signal.

That's all the information i can provide.

Yours,

Frank Everdij