tudelft-iv/multi_sensor_calibration #3: Segmentation fault,Core Dump

Rajkumarpalanisamy96, 2019-09-13:
Hi,
I am using RSlidar instead of Velodyne and i am using a convertor ([https://github.com/EpsAvlc/rslidar_to_velodyne]) after conversion i tried running it i got segmentation fault again and after carefull debugging i was able to find where the issue is its in keypointdetector .
I am not sure what type of data is it expecting and what is the issue .

Rajkumarpalanisamy96, 2019-09-13:
I am having issues at this line 
`		pcl::PointCloud<pcl::PointXYZ> pattern = keypointDetection(cloud, config_);`

Rajkumarpalanisamy96, 2019-09-13:
I changed the new converted point cloud in the sub point_cloud_subscriber_

RonaldEnsing, 2019-11-13:
A velodyne point type is used. This is not a standard PCL point type. This point type is defined in the velodyne driver ros package. To avoid a dependency on the velodyne package only for this point type, it is defined in the multi sensor calibration package as well. The velodyne point type contains information about the ring number associated with the point and this information is used by the detection algorithm. If you use a conversion, then you will thus need to convert to this point type accordingly.

farashimca, 2020-06-15:
Even if you use RSlidar, the Velodyne structure attribute remain same in other type of lidar as well .