tudelft-iv/multi_sensor_calibration #4: single camera / single radar extrinsic calibration

cedros23, 2019-09-18:
Hello,

Thanks for sharing the nice work.

I have a question regarding the system capabilities.
We don't have lidar in our vehicle, so I would like to focus on radar to camera extrinsic estimation.

When I check the custom calibration board, it doesn't have the checkerboard pattern with radar reflector. Rather, it has the four black holes for the lidar returns combined with radar reflector. Would the same custom calibration board work with mono camera & radar?

We have the possibility to use stereo system as well, so I would like to know capabilities without lidar.

Thanks in advance,
Kind Regards,
Ugur

RonaldEnsing, 2019-09-18:
We have not done experiments with the combination mono/stereo camera and radar (without lidar). I would think that it's possible. It should not be a lot of work to change the configuration files in order to perform such a calibration. If you are going to proceed doing that, please share your findings by commenting in this issue.

ixtiyoruz, 2019-12-12:
> Hello,
>
> Thanks for sharing the nice work.
>
> I have a question regarding the system capabilities.
> We don't have lidar in our vehicle, so I would like to focus on radar to camera extrinsic estimation. [...]

Hi, have you been able to do the calibration with radar && camera only?

cedros23, 2019-12-13:
Hello, unfortunately we could not give it a go, busy with other things.

ixtiyoruz, 2019-12-13:
> Hello, unfortunately we could not give it a go, busy with other things.

It is ok, thanks for the response. If you are able to do that,  i would be very glad if you let me know how .

ishfaq-gist, 2019-12-25:
> > Hello, unfortunately we could not give it a go, busy with other things.
> 
> It is ok, thanks for the response. If you are able to do that, i would be very glad if you let me know how .

Hi, have you been able to do calibration with radar and camera?

ixtiyoruz, 2019-12-25:
> > > Hello, unfortunately we could not give it a go, busy with other things.
> > 
> > 
> > It is ok, thanks for the response. If you are able to do that, i would be very glad if you let me know how .
> 
> Hi, have you been able to do calibration with radar and camera?

Hi, not yet i also been busy with another tasks.

ishfaq-gist, 2019-12-26:
> > > > Hello, unfortunately we could not give it a go, busy with other things.
> > > 
> > > 
> > > It is ok, thanks for the response. If you are able to do that, i would be very glad if you let me know how .
> > 
> > 
> > Hi, have you been able to do calibration with radar and camera?
> 
> Hi, not yet i also been busy with another tasks.

Thank you. Please share if you find something .

ixtiyoruz, 2020-05-11:
> Thank you. Please share if you find something .

You can calibrate radar and camera by manually giving 3d lidar coordinates, directly calibration doesn't work well.

ishfaq-gist, 2020-07-19:
Hi, 

> You can calibrate radar and camera by manually giving 3d lidar coordinates, directly calibration doesn't work well.

hi, First of sorry for being late. Have you done the task with only camera and lidar?

EaiPool, 2021-03-15:
Speaking of late, 

The package no longer requires exactly 1 lidar, radar and camera. As long as you set the reference sensor, your radar+camera sensor setup should now be calibratable.

dkhanna511, 2021-07-19:
Hi @EaiPool. Is it possible to calibrate 2 cameras with a single LIDAR keeping LIDAR as a reference sensor? It would be a great help if you could validate this.

EaiPool, 2021-07-19:
Yeah, that should be doable. The only requirement is that the boards detected by each camera are also detected by the reference sensor. 

Some of the registration methods will also require that boards detected by one camera are detected by the other, but not all registration methods need this.

dkhanna511, 2021-07-19:
Thanks! I'll get back if this is done or I face some issue.