tudelft-iv/multi_sensor_calibration #7: data loading

ixtiyoruz, 2019-12-12:
Hi, thank you for your open-source. 
(I am new in the field of Radar and multi Sensor calibration). I have one question. I assume that you first recorded images, radar information, and lidar information simultaneously and stored them. So, I wonder how are you reading them, where they should be stored?

EaiPool, 2021-03-17:
Hey,

In case you're still on this: you should use the radar detector to extract the calibration board detections. The data of these detections is what gets stored in the CSV files. 
Depending on what you mean by "raw radar data", you should either create a ROS node to convert into the format used by the radar detector, or write something which uses the non-ROS radar detection backend to extract correct detections from your raw data.

ixtiyoruz, 2021-03-17:
Hi, thank you for your response. I wrote the question out of poor knowledge, however now I have a clue of what to do. Than you very much.

closed 2021-03-17 by ixtiyoruz