tudelft-iv/multi_sensor_calibration #8: lidar_detector_node pcl warning

ishfaq-gist, 2019-12-21:
hi. 
we made the calibration board according to the same dimension as given in code. But we are getting the following error as attached in files...  we are using **"velodyne-32 channel"**  

In the lidar_detector_node giving us the following error. 

"ring too small"

and nothing is showing in the topic is publishing by this detector

[image]

In the mono_detector_node giving us the following exception

[image]
"Number of circles found: '0', but should be exactly 4."

your help is requested please...

dejongyeong, 2020-06-08:
@ishfaq-gist  I came across the same issue on ring too small. Would like to know what is the issue of that? Thanks.

ishfaq-gist, 2020-06-09:
@dejongyeong I think the problem will be solved if you try to change the location of the calibration board from the vehicle (I think now your board is far from the sensor  e.g more than 5m)  or your circles front and backside is different than the desired.

closed 2020-06-17 by ishfaq-gist