tudelft-iv/multi_sensor_calibration #10: Question on Calibration Board Pattern

dejongyeong, 2020-05-19:
Hi Guys,

The sensors that were used are mono camera, lidar and radar. I would like to ask about is the proposed pattern be used for mono camera and would like to ask about the size of the trihedral corner reflector.

Thanks.

RonaldEnsing, 2020-05-19:
More info on the calibration board can be found here: https://github.com/tudelft-iv/multi_sensor_calibration/blob/master/docs/calibration_board.md

Does this answer your question about the trihedral reflector: https://github.com/tudelft-iv/multi_sensor_calibration/issues/2 ?

dejongyeong, 2020-05-19:
Hi @RonaldEnsing, Thanks for getting back to me.

Which means for the board pattern, I doesn't have to incorporate a checkerboard pattern into the current board pattern right?

Thanks for the link to the question about the trihedral reflector. Just wondering what radar sensor did you used on this? Currently, we are using SmartMicro Automotive UMRR T-153 radar.

Thanks.

frankeverdij, 2020-05-20:
Hello dejongyeong,
The radr used is also mentioned in issue #2

RonaldEnsing, 2020-05-20:
You do not need to include a chessboard pattern like this [1] if that is what you mean.

[1] https://docs.opencv.org/2.4/doc/tutorials/calib3d/camera_calibration/camera_calibration.html#results

dejongyeong, 2020-05-20:
@RonaldEnsing @frankeverdij Thanks for the response. Much appreciated!!!

closed 2020-05-20 by dejongyeong