tudelft-iv/multi_sensor_calibration #14: radar_converter node missing

dejongyeong, 2020-07-06:
Hi, just noticed that the radar_converter node is missing from the package. Does anyone has the solution to that? Thanks.

EaiPool, 2021-03-15:
Hello,

Our internal radar sends out a homebrew ROS message format. We did not want to make this repo dependent on that message format, so we wrote a conversion node that we can use internally. The launch file accidentally referred to that conversion node, and this has been removed in the latest update. If you wish to use this with your own radar and it sends out a custom message format, you will have to write a conversion node for that yourself.

This is somewhat of a general issue within ROS, as there is not really a standard radar format yet. However, there is one in the making (see [here](https://discourse.ros.org/t/radar-messages-standard/14355)), so perhaps it will soon make sense for us to change over to that format which might simplify things for you as well.

closed 2021-03-15 by EaiPool

dejongyeong, 2021-03-15:
Hi @EaiPool, thanks for the insightful information. A quick question here regarding the radar message type. Would it be possible to directly utilise the output message (see [link](https://autonomoustuff.atlassian.net/wiki/spaces/RW/pages/17509820/Delphi+ESR)) from AutonomouStuff ESR 2.5 Radar after remapping the topic name?

[image]

Thanks for the update once again.

EaiPool, 2021-03-15:
Assuming that `raadar_msgs` is a typo, then yes, I would expect that to be directly compatible. I believe they might be the maintainer for the radar message format that we're using, if I'm not mistaken. See their repo [here](https://github.com/astuff/astuff_sensor_msgs).

dejongyeong, 2021-03-15:
Thanks for the information. Much appreciated it.