tudelft-iv/multi_sensor_calibration #16: Mono-camera and radar only

CRalli1, 2020-11-30:
Hi, 
Wondering if this can be used without Lidar or stereo cameras?
Also wondering if it can be used with raw radar data rather than a point cloud?
Cheers

ishayniart, 2020-12-17:
join to this question. is it possiable?

EaiPool, 2021-03-15:
The current update allows you to work with an arbitrary number of sensors, so working without lidar or stereo camera should be fine. However, we have only tested this with setups in which  the reference sensor was able to see all detections, so I cannot say for sure if it works if that is not the case. 

Additionally, some optimization methods do not work with more than one 2D radar.

Finally, I am not sure what you mean with "raw radar data". What does this refer to, exactly?

CRalli1, 2021-03-16:
Thanks for your response. That looks very useful. The TI AWR range offers a secondary PCB that allows you to access the data directly from the ADC and they call this the 'raw data'. Since asking this my project has moved forward and I am no longer concerned about this  functionality. Thanks for your help,
Charlie

closed 2021-03-16 by CRalli1

fabioreway, 2021-11-07:
We had some issues with the pose estimation while performing a mono camera + radar only calibration. Our radar appears to be on top of the camera, when the opposite is true.

Has anyone successfully done it?