tudelft-iv/multi_sensor_calibration #18: Problem with optimization function "error: Parsing coordinates for the following line"

mounchiliu, 2021-02-20:
Hi there, I got a problem when I tried to run the optimization node with the example data provided.
I run the order as the README.md file illustrated, but I got problem when run

- python3 src/main.py --lidar data/example_data/lidar.csv --camera data/exampl
e_data/camera.csv --radar data/example_data/radar.csv --calibration-mode 3 --visualise

or use rosrun to setup the optimization node
- rosrun optimization server.py

How can I solve this problem?

mounchiliu, 2021-02-22:
I suppose this problem may be caused by the different version of python.
I solved it by:
- updating python3.5 to python3.7 and set python3.7 as the default python3 version of the system
- sudo pip3.7 install rmsd

Then, I rerun the optimization method and got the following problem. I have also attached the solution to each problem as below.

- ModuleNotFoundError: No module named 'sklearn'
 1. start a new terminal
 2. pip3 install scikit-learn

- ModuleNotFoundError: No module named 'yaml'
 1. pip3 install pyyaml

- RuntimeError: matplotlib2tikz has been renamed to tikzplotlib <https://pypi.org/project/tikzplotlib>
1. pip3 install tikzplotlib
2. open file that got error, replace 'matplotlib2tikz ' with 'tikzplotlib'. Save the file.
3. rerun the optimization method to get the result. :)

mounchiliu, 2021-02-22:
I also got a problem
- ".error:transport error completing service call: unable to receive data from sender, check sender's logs for"

This can be solved by add: 

> **add_definitions(-DEIGEN_DISABLE_UNALIGNED_ARRAY_ASSERT)**

to both Cmakelists.txt of accumulator and common profile.

At last, re-catkin_make the project.

EaiPool, 2021-03-15:
Hey there, 
You are correct on the issue of tikzplotlib, this has been corrected in the latest version.

Many of your other issues (python version, missing sklearn and yaml) can be mitigated by, as you found, installing the dependencies as mentioned in the readme. 

This latest issue, however, I cannot reproduce. Can you verify that it does not happen when you have all the dependencies properly installed? Otherwise, can you give more information regarding when it happens? During compilation, runtime, what line, your OS, etc?

mounchiliu, 2021-03-16:
> Hey there,
> You are correct on the issue of tikzplotlib, this has been corrected in the latest version.
>
> Many of your other issues (python version, missing sklearn and yaml) can be mitigated by, as [...]

Hi, I am using ubuntu 16 with  ROS Kinetic.
I suppose the issue caused because the default python version provided by system is python 2.7 and python 3.5, which may not compatible with the latest **rmsd** lib.  When I tried to use the library **rmsd** with python 3.5, it shows the error I mentioned above. Then, I tried python 3.7, and it works.

mounchiliu, 2021-03-16:
> I also got a problem
> 
> > error:transport error completing service call: unable to receive data from sender, check sender's logs for
> 
> This can be solved by add the following to the both Cmakelists.txt of accumulator and common profile:
> 
> > **add_definitions(-DEIGEN_DISABLE_UNALIGNED_ARRAY_ASSERT)**
> 
> > catkin_make

The second problem also comes with the error
`Assertion failed: (reinterpret_cast<size_t>(array) & 0xf) == 0 && "this assertion is explained here: " .....`

I only got the problem when I was using x86 system to run the code.
I also tried the code on x64 system. The code will not show this kinda problem on x64(amd64) system.

EaiPool, 2021-03-17:
>The second problem also comes with the error
Assertion failed: (reinterpret_cast<size_t>(array) & 0xf) == 0 && "this assertion is explained here: " .....

Can you please put down the entire error log? I cannot really assess this problem without the file and line number where the compilation error occurs. 

>I suppose the issue caused because the default python version provided by system is python 2.7 and python 3.5, which may not compatible with the latest rmsd lib. When I tried to use the library rmsd with python 3.5, it shows the error as I mentioned above. Then, I tried python 3.7, and it works.

I found one case of f-string formatting which indeed made it impossible to run with older versions of python. However, with that changed, I was able to run the optimization without any problem. As such, I do not think the issue here is rmsd. If you still experience issues with python 3.5, please post the entire error that you get.