tudelft-iv/multi_sensor_calibration #19: 2d lidar and depth camera?

kaushikbalasundar, 2021-03-17:
Can this be used to calibrate a 2d Lidar and depth camera with overlapping FOV?

EaiPool, 2021-03-17:
It all depends if you can either 1) use one of the included detectors for detect the four circles or 2) build your own detector to do so.

The information present in a depth camera is I think equal or at least similar to that of a stereo camera setup. You can either use the included stereo detector for that, or modify it so it works with your inputs.
For the 2D lidar, however, I doubt if it's possible. I do not know how exactly you mean it is 2D, but my assumption is that it's essential a single-plane lidar. With such a lidar, it will not be possible to detect all four circles on the calibration board (even detecting the 3D position of the center point of a circle might already be impossible), which makes it unusable for this calibration package.

kaushikbalasundar, 2021-03-17:
Alright, makes sense. Thanks!

closed 2021-03-17 by kaushikbalasundar