tudelft-iv/multi_sensor_calibration #21: Lidar converter

ffent, 2021-04-07:
Hi, 

I really want to thank you for your great work. I'm currently working on a lidar converter to use your code with a lidar sensor different from a Velodyne sensor and would like to ask for some further information. Could you please provide the PointField message as part of the PointCloud2 message of the /velodyne_points ROS topic, which the lidar_detector is subscribing to?

I'm currently working with the following PointField message, but getting errors from the lidar detection node.
<details>
<summary>sensor_msgs/PointField</summary>
<br>

```
fields:
    -
        name: "x"
        offset: 0
        datatype: 7
        count: 1
    -
        name: "y"
        offset: 4
        datatype: 7
        count: 1
    -
        name: "z"
        offset: 8
        datatype: 7
        count: 1
    -
        name: "intensity"
        offset: 12
        datatype: 7
        count: 1
    -
        name: "ring"
        offset: 16
        datatype: 4
        count: 1
```

</details>
<br />

EaiPool, 2021-04-09:
You should be alright with these fields: all that's required is x, y, z and ring (see [here](https://github.com/tudelft-iv/multi_sensor_calibration/blob/master/lidar_detector/include/lidar_detector/keypoint_detection.hpp#L51-L53)). 

The way the pcl stuff works is kind of obtuse though, so if you need more help please post the exact error message that you get.
I also see you've opened a new issue, please let me know if that one supercedes this one.

ffent, 2021-04-09:
Thanks @EaiPool for your quick response. The PointCloud2 message seems not to be the issue but rather the configuration of the lidar detector. I opened a new issue (#22) regarding this topic and would be very thankful if you could help me out on this.

closed 2021-04-09 by ffent

avinash-yadav1, 2021-12-11:
@EaiPool  @ffent  currently i'm using Ouster lidar and i tried to run the Lidar detector node with it unfortunately i'm getting an error: **Failed to find match for field 'ring'.** Please help me on this how to resolve it.