tudelft-iv/multi_sensor_calibration #27: Single Lidar Multi Camera calibration

dkhanna511, 2021-07-16:
Hi, since in the paper, multiple modalities have been mentioned, is it possible to use this repository for calibrating multiple cameras with a single lidar. Or it works only for 1 combination, ie, a LIDAR,a camera, and a Radar.

RonaldEnsing, 2021-07-16:
Any combination of modalities should work, but we haven't tested anything else than 1 radar, 1 lidar and 1 stereo camera as far as I know.

dkhanna511, 2021-07-19:
okay. I'll try it out. Thanks!

paaraujo, 2022-05-03:
I've tried 1 LiDAR and 1 stereo camera and I got a fatal error when I call the optimizer. It seems that the code expects radar detections and it just stops if none is detected. I didn't play with the code, but I think it is possible to turn off the radar optimization to calibrate the camera and the LiDAR.