tudelft-iv/multi_sensor_calibration #31: Failed to find match for field 'ring'.

avinash-yadav1, 2021-12-11:
Hi i'm using Ouster lidar and on running the lidar with ouster point cloud i'm getting error: **Failed to find match for field 'ring'.** 
Ouster is publishing PointCloud2 message on running rostopic echo command i can see there is a field 'ring' available in the cloud. Anyone please help me on this issue.

Rob5200, 2021-12-16:
Hi, did you change the topic name in the lidar_detector/src/lib/node_lib?
If so, maybe the issue lays in the name of the ring field, or the type. In my case the type of the ring field was uint8_t.

To change the type, go to: "/lidar_detector/include/lidar_detector/keypoint_detection.hpp"
and change;

namespace Lidar {
	struct PointWithDist {
		PCL_ADD_POINT4D; ///< quad-word XYZ

		float distance; ///< place to store the distance
		uint16_t ring; ///< laser ring number
		EIGEN_MAKE_ALIGNED_OPERATOR_NEW ///< ensure proper alignment
	} EIGEN_ALIGN16;
}

// Note that the following does not map to the float 'distance', as that is written to internally.
POINT_CLOUD_REGISTER_POINT_STRUCT(
	Lidar::PointWithDist, (float, x, x) (float, y, y) (float, z, z)
		(uint16_t, ring, ring));

namespace lidar_detector {

/// Main function to delete calibration board pose from lidar (velodyne) point cloud
pcl::PointCloud<pcl::PointXYZ> keypointDetection(pcl::PointCloud<Lidar::PointWithDist> const & cloud, Configuration const & config);

}

To: 

namespace Lidar {
	struct PointWithDist {
		PCL_ADD_POINT4D; ///< quad-word XYZ

		float distance; ///< place to store the distance
		uint8_t ring; ///< laser ring number
		EIGEN_MAKE_ALIGNED_OPERATOR_NEW ///< ensure proper alignment
	} EIGEN_ALIGN16;
}

// Note that the following does not map to the float 'distance', as that is written to internally.
POINT_CLOUD_REGISTER_POINT_STRUCT(
	Lidar::PointWithDist, (float, x, x) (float, y, y) (float, z, z)
		(uint8_t, ring, ring));

namespace lidar_detector {

/// Main function to delete calibration board pose from lidar (velodyne) point cloud
pcl::PointCloud<pcl::PointXYZ> keypointDetection(pcl::PointCloud<Lidar::PointWithDist> const & cloud, Configuration const & config);

}

avinash-yadav1, 2021-12-20:
Thanks @Rob5200  for the suggestion.

closed 2021-12-20 by avinash-yadav1